d-a-v/ESPWebDAV #5: [FEATURE REQUEST]ESP32 support

luc-github, 2021-01-12:
I know it may need some esp32 core update but any chance for ESP32  support ?

d-a-v, 2021-01-12:
There are [little updates](https://github.com/esp8266/Arduino/pull/7459) done in esp8266 arduino core to enable web hooks.
Unfortunately the PR is mixed with other internal changes.
Functionality main changes are:

- [`WEBSERVER_HAS_HOOK`](https://github.com/esp8266/Arduino/pull/7515)
- [definitions](https://github.com/esp8266/Arduino/pull/7459/files#diff-799c46c81de7a8939be0ec4f4aef5ad592378592d337950447996243b78f37e0R96-R98)
- [declaration](https://github.com/esp8266/Arduino/pull/7459/files#diff-799c46c81de7a8939be0ec4f4aef5ad592378592d337950447996243b78f37e0R294)
- [`addHook()`](https://github.com/esp8266/Arduino/pull/7459/files#diff-799c46c81de7a8939be0ec4f4aef5ad592378592d337950447996243b78f37e0R219-R231)
- [hook call](https://github.com/esp8266/Arduino/pull/7459/files#diff-d860dce5ac3b563de84de3b93268575a24cc7afa5794e4eeadb33646b454f680R89-R95)
- [example](https://github.com/esp8266/Arduino/blob/34545a160da7c835aa87d7bace62192aee50fa78/libraries/ESP8266WebServer/examples/HelloServer/HelloServer.ino#L89-L145)

I guess the hook call in ESP 32 arduino would have to happen [there](https://github.com/espressif/arduino-esp32/blob/b0e896e9ae7df971de333307c5bc9907cc81a0a8/libraries/WebServer/src/Parsing.cpp#L98).

Do you think you could propose it ?

luc-github, 2021-01-12:
So if I implement each steps - in theory it should be ok ?
I give a try and report to you ^_^ thank you

d-a-v, 2021-01-12:
> So if I implement each steps - in theory it should be ok ?

It *will* be OK, no reason against it.

I forgot to mention the return type in 
```cpp
bool WebServer::_parseRequest(WiFiClient& client)
```

that must be changed to `ClientFuture`.
- `return false` => `return CLIENT_MUST_STOP`
- `return true` => `return CLIENT_REQUEST_CAN_CONTINUE`
- update the caller of this function to [manage the return value (4 choices)](https://github.com/esp8266/Arduino/pull/7459/files#diff-89e77d8bf4b9262a6819cf079785890f77dc7e8c62cf65d9e2e05f3d2a27a82aR331-R357).

luc-github, 2021-01-12:
ok noted - thank you 👍

luc-github, 2021-01-15:
@d-a-v I did first part https://github.com/luc-github/arduino-esp32/commit/b5233795d7e61a2fc724397bb61db7065921bb41
seems working - I have followed your instructions even I have no idea of what I am doing ^_^

Testing with HelloServer.ino

`/` and I got `hello from esp32!` in browser=>ok
`/dump` and no answer in browser =>ok
`/yes` and got 
```
File Not Found

URI: /yes
Method: GET
Arguments: 0
```
On serial
```
Connected to WIFI_OFFICE_A2G
IP address: 192.168.1.61
MDNS responder started
HTTP server started
A useless web hook has passed
A useless web hook has passed
The dumper web hook is on the run
(<32> chars)Host: 192.168.1.61
Connection: (<32> chars)keep-alive
Upgrade-Insecure-Req(<32> chars)uests: 1
User-Agent: Mozilla/5.(<32> chars)0 (Windows NT 10.0; Win64; x64) (<32> chars)AppleWebKit/537.36 (KHTML, like (<32> chars)Gecko) Chrome/87.0.4280.88 Safar(<32> chars)i/537.36
Accept: text/html,appl(<32> chars)ication/xhtml+xml,application/xm(<32> chars)l;q=0.9,image/avif,image/webp,im(<32> chars)age/apng,*/*;q=0.8,application/s(<32> chars)igned-exchange;v=b3;q=0.9
Accep(<32> chars)t-Encoding: gzip, deflate
Accep(<32> chars)t-Language: fr-FR,fr;q=0.9,en-US(<19> chars);q=0.8,en;q=0.7

Telling server to forget this connection
```
Now checking WebDav code but I have realized that FS of ESP32 is different from ESP8266 so cannot be drop as it is - I will check that part first before pushing PR to ESP32 core

d-a-v, 2021-01-15:
The "hooked" way of using espwebdav imposes that the files on the filesystem are stored in `/theHookedDirectory/`.
I understand that it is currently a limitation. 
Until this is improved, can you try to store files in the FS's directory `/yes/` and check if you can see them in a webdav browser ?

luc-github, 2021-01-15:
Sorry you misunderstood my test 
/ is to check root
/dump is the hook 
`/yes` was to go to not found handler so it is working as expect

luc-github, 2021-01-16:
hi here the POC https://github.com/luc-github/ESPWebDAV/commit/d949d6fe47a53c1e472af5bcd278bd54d3b3ffbd

I have some refresh problem - but not sure how to address it

luc-github, 2021-01-17:
OK I have followed your advice and indeed debug was part of issue
the other part was the File system difference SPIFFS esp32 is different to SPIFFS
* the file/dir detection - I use exists instead of open
* the mkdir feature - it does not work

d-a-v, 2021-01-17:
> the mkdir feature - it does not work

spiffs has no support for directories, but sdfs has. What FS did you try with ?
Does esp32/arduino have littlefs (it has support for directories too).

luc-github, 2021-01-17:
yes I know, I am trying with SPIFFS - ESP32 has SPIFFS/FAT /SD not yet LITTLEFS , it is planned in 2.0 but currently it can be used as external FS - I will try it later

I also suspect the pb with the hooked is also related with directory pb, a nice to have would be to separate the url from the real directory name 
for example use http://192.168.1.61/dav but map the / of the FS or /www/dav of FS

d-a-v, 2021-01-17:
> for example use http://192.168.1.61/dav but map the / of the FS or /www/dav of FS not /dav

That is a fair request

luc-github, 2021-01-18:
still working on difference of FS
here as example the output between the 2 littleFS

ESP8266
````
>>>>>>>> dir /
1.PNG                                      0MiB     76KiB 77569
1.md                                       0MiB      0KiB 6
[Z]
[dav]
    0.txt                                      0MiB      0KiB 0
    1.txt                                      0MiB      0KiB 0
    2.txt                                      0MiB      0KiB 0
    Capture.PNG                                0MiB     76KiB 77569
    espconf.ini                                0MiB      2KiB 2239
    readme.md                                  0MiB      0KiB 6
<<<<<<<< dir
interactive: F/ormat D/ir C/reateFile
Heap stats: free heap: 46536 - max block: 46296 - fragmentation: 1%
interactive: F/ormat D/ir C/reateFile
Heap stats: free heap: 46536 - max block: 46296 - fragmentation: 1%
````

EPS32
```
>>>>>>>> dir /
/readme.md                                 0MiB      0KiB 6
/test.file                                 0MiB      0KiB 0
[/dav]
    ///dav/New Text Document.txt               0MiB      0KiB 0
    ///dav/www.txt                             0MiB      0KiB 0
<<<<<<<< dir
interactive: F/ormat D/ir C/reateFile
Heap stats: free heap: 356360
interactive: F/ormat D/ir C/reateFile
Heap stats: free heap: 356360
```

luc-github, 2021-01-20:
The simple_esp32.ino is ok 
The hooked_esp32.ino is still failing
[esp8266.log](https://github.com/d-a-v/ESPWebDAV/files/5845106/esp8266.log)
[esp32.log](https://github.com/d-a-v/ESPWebDAV/files/5845107/esp32.log)
Esp32 do not have HTTP_HEAD neither keepAlive connection - so I think I have added them  but I may still miss one part to make the connection to succeed
here the esp32 webserver  modified https://github.com/luc-github/arduino-esp32/commits/webserver-hook/libraries/WebServer

I do not know what is missing 😭

agoy507, 2021-03-27:
How wiring sdcard to esp32 ?
what pin configuration on sourcecode?
can you describe ?

d-a-v, 2021-03-27:
SD cards have mmc emulation. It's on the SPI bus.
You need one of the SPI buses (MISO, MOSI, CLK) and another unused GPIO for CS.

agoy507, 2021-03-29:
thanks :D