d-a-v/ESPWebDAV #13: conflict with <SdFat.h>

bvsall, 2021-03-28:
I am trying to make WiFi SD for 3D Printer on ESP32 using your ESPWebDAV library.
For SD I use <SdFat.h> because I think it's faster.
If i connect (uncomment) your library, an error occurs during compilation.
I'm not very good at C. Can you tell me what i can do?
To make it easier to understand, I removed everything unnecessary from the program.

======================================================================================
#include <ESPWebDAV.h>
#include <SdFat.h>
SdFat SD;
File myFile;

void setup() {

  if (!SD.begin(5, SD_SCK_MHZ(20))) {
    Serial.println("initialization failed!");
    return;
  }
  Serial.println("initialization done.");

  // open the file.
  myFile = SD.open("test.txt", O_RDONLY);

  myFile.close();
}
void loop() {
  // nothing happens after setup
}

=======================================================================================

C:\Users\BvS\Documents\Arduino\sketch_mar28a\sketch_mar28a.ino: In function 'void setup()':
sketch_mar28a:16:10: error: no match for 'operator=' (operand types are 'fs::File' and 'File32')
   myFile = SD.open("test.txt", O_RDONLY);
          ^
In file included from C:\Users\BvS\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.5\libraries\ESPWebDAV-2/ESPWebDAV.h:86:0,
                 from C:\Users\BvS\Documents\Arduino\sketch_mar28a\sketch_mar28a.ino:2:
C:\Users\BvS\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.5\libraries\FS\src/FS.h:47:7: note: candidate: fs::File& fs::File::operator=(const fs::File&)
 class File : public Stream
       ^
C:\Users\BvS\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.5\libraries\FS\src/FS.h:47:7: note:   no known conversion for argument 1 from 'File32' to 'const fs::File&'
C:\Users\BvS\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.5\libraries\FS\src/FS.h:47:7: note: candidate: fs::File& fs::File::operator=(fs::File&&)
C:\Users\BvS\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.5\libraries\FS\src/FS.h:47:7: note:   no known conversion for argument 1 from 'File32' to 'fs::File&&'
Несколько библиотек найдено для "ESPWebDAV.h"
 Используется: C:\Users\BvS\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.5\libraries\ESPWebDAV-2
Не используется: C:\Users\BvS\Documents\Arduino\libraries\ESPWebDAV
Несколько библиотек найдено для "WiFi.h"
 Используется: C:\Users\BvS\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.5\libraries\WiFi
Не используется: C:\Program Files (x86)\Arduino\libraries\WiFi
Несколько библиотек найдено для "SdFat.h"
 Используется: C:\Users\BvS\Documents\Arduino\libraries\SdFat
Не используется: C:\Users\BvS\Documents\Arduino\libraries\SdFat-master
Не используется: C:\Users\BvS\Documents\Arduino\libraries\SdFat_-_Adafruit_Fork
exit status 1
no match for 'operator=' (operand types are 'fs::File' and 'File32')

luc-github, 2021-03-28:
The SDFat API is not same as ESP32 SD API - so it won't work by just changing the SD reference

You need to virtualize SDFat to use same FS API as ESP32 SD

pipi61, 2021-09-22:
try the original SDMMC directory, I tested it on esp32cam
...
//#include <SPIFFS.h>
//#include <LITTLEFS.h> //pipi
#include <SD_MMC.h>
...
//FS& gfs = LITTLEFS; 
//#define FILESYSTEM LITTLEFS
//#define FILESYSTEM SPIFFS
FS& gfs = SD_MMC;
#define FILESYSTEM SD_MMC