d-a-v/ESPWebDAV #14: Redirect "/" to "/.www/index.html"

idolpx, 2021-06-21:
I want to use "/" as root for WebDAV but I would like to rewrite the URL for "/" to "/.www/index.html" so my web configuration app will be accessible by default.

```
#if WEBSERVER_HAS_HOOK
WebServer::HookFunction hookWebDAVForWebserver(const String& davRootDir, ESPWebDAVCore& dav)
{
    return [&dav, davRootDir](const String & method, const String & url, WiFiClient * client, WebServer::ContentTypeFunction contentType)
    {
        if (url == "/")
        {
            DBG_PRINT("WE ARE AT ROOT!");
            url = "/.www/index.html";
        }

        if (url.indexOf(".www") != 0)
        {
            DBG_PRINT("CLIENT_REQUEST_CAN_CONTINUE, %s is not seen in %s", davRootDir.c_str(), url.c_str());
            return WebServer::CLIENT_REQUEST_CAN_CONTINUE;
        }
```
This does not work.  Is there a way to do this?

idolpx, 2021-06-21:
Actually it would be cool if it could detect a webbrowser and just bypass webdav and serve up web pages as usual.  That's kinda what I'm trying to do here.  

I love having the webdav access to root but when accessing via a web browser I just want it to serve up html.

d-a-v, 2021-09-09:
@idolpx The API and the example [are updated](https://github.com/d-a-v/ESPWebDAV/blob/0eb4e3165e9506d4cccecdf635b8a663be86e755/examples/Hooked/Hooked.ino#L141-L151) with what's needed about your feature request.

d-a-v, 2021-12-02:
Have you had a chance to try ?