d-a-v/ESPWebDAV #16: No documentation. How can i use SD card with FAT instead of LittleFS ????

nightgryphon, 2021-08-03:
How can i replace LittleFS with FAT SD card with SPI access? Can i return back SDFat used within original code or there is another solution specific for this library? How can i turn storage on and off to change SD card or share card access?

Unlike original quite self explaining ESPWebDAV code this library is complete mess of callbacks and such stuff without a single line of comments or any usable documentation. All the "documentation" is regarding mounting webdav at other OS instead of library code documentation. Please help :(

d-a-v, 2021-08-04:
If this is too much a mess you can still use the other one.
I use it with sd. Juste replace littlefs with sdfs.
You can also use the other's repo examples.
I'll try to provide an example but not before a few weeks.
The callbacks are meant to share the port 80 with esp8266 webserver. They are not necessary for a start with SD.

nightgryphon, 2021-08-06:
unfortunately i can't make that work with replacing LettleFS to SDFS. It produce a huge list of errors and does not compile.

Arduino 1.8.15 @ win 8.1

Board:
ESP8266 Community 3.0.2 -> "Lolin(WeMos) D1 r2 & mini"

Libraries are insalled with arduino library manager:
ESPWebDav 2.1.2
SdFat 2.0.7

edited 'Simple' sample:
`#include <ESP8266WiFi.h>
#include <ESP8266mDNS.h>
//#include <LittleFS.h>
#include <SdFat.h>
#include <SDFS.h>
#include <ESPWebDAV.h>

.............

//FS& gfs = SPIFFS;
//FS& gfs = LittleFS;
FS& gfs = SDFS;
`

Errors
`In file included from C:\Users\Temp\AppData\Local\Temp\arduino_modified_sketch_950713\Simple.ino:55:
C:\Users\Temp\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\libraries\SDFS\src/SDFS.h:232:5: error: 'sdfat' does not name a type
  232 |     sdfat::SdFat* getFs()
      |     ^~~~~

......... bunch of errors related to that first one ............
`

pipi61, 2021-09-22:
try the original SDMMC directory, I tested it on esp32cam
...
//#include <SPIFFS.h>
//#include <LITTLEFS.h> //pipi
#include <SD_MMC.h>
...
//FS& gfs = LITTLEFS;
//#define FILESYSTEM LITTLEFS
//#define FILESYSTEM SPIFFS
FS& gfs = SD_MMC;
#define FILESYSTEM SD_MMC

RPdenBoer, 2022-11-05:
@pipi61 

Hi, did you make any other changes than what you mentioned here? I'm trying to use SD_MMC also (1 bit mode) but getting a lot of compile errors. Should this work fairly easily or is it a bit involved to get going?
Thanks for your help.

pipi61, 2022-11-05:
I use only 4 bit mode

heinerolli, 2023-02-01:
@d-a-v  Thank you for providing this 'enhanced' ESPWebDav system. I got it to run with the std. example using LittleFS, however, I would like to use SD card....
-> Could you please provide an example based on your library?
(I did try the 'simple' example from ardyesp, but I get a compile error:
```
Simple:115:11: error: 'class ESPWebDAV' has no member named 'init'
  115 |   if(!dav.init(SD_CS, SPI_FULL_SPEED, 80))   {
```
All the best.