d-a-v/ESPWebDAV #17: Timeout never expires

pipi61, 2021-09-22:
Hi!
in ESPWebDAV.cpp this good?
...
1502    m_persistent_timer_ms = millis();
1503
1504    m_persistent = ((millis() - m_persistent_timer_ms) < m_persistent_timer_init_ms);
...
this timer never expires...
----------------------------------------------------
another suggestion: 
the buffer size is very small
8899     char buf[128]; /// XXX use stream::to(): file.to(client);
and
1013         uint8_t buf[128];
if it changes to a higher value (I tried 1440 * 3), the transfer rate will increase significantly !!!

Thank You!
Istvan

d-a-v, 2021-12-02:
Thanks for the suggestions !
I missed this notification.
>  this timer never expires...

I'll have to check

>  the buffer size is very small

True. In the meantime we have (on esp8266) `file.sendAll(client)`.
On esp32, the buffer can be increased, but storing 1440*3 bytes on stack is way too large.
Did you try with simply with TCP_MSS (1440) ?

d-a-v, 2023-01-17:
Did you try the git version of this library (one year old) ?
Transfer had been optimized in the latest commits.

pipi61, 2023-01-17:
Hi!
I modified, flashed and am using 20210919 version 2.1.3
I use data transfer to a 3D printer.
Removed the TFCARD from the printer, insert it into the ESP32CAM. The TF socket shares over the resistor network and the flat cable.
Normally the ESP32 is power on, but in reset mode, the printer uses the TFCARD.
If esp32 is power on, and works, then esp32 will use TFCARD and transfer 3d files on webdaw (in this case the printer does not see it)
Thank You! Great works :)
[image]

pipi61, 2023-01-17:
[image]
[image]

d-a-v, 2023-01-20:
Thanks for the report !

closed 2023-01-20 by d-a-v