d-a-v/ESPWebDAV #21: [feature request] virtual directory tree for windows explorer clients

nightgryphon, 2022-07-16:
Unfortunately windows explorer client can not access davRootDir without check availability for all upper path.
If I set davRootDir to anything except "/" the windows explorer can not access that path even using direct link with all path included
`\\<IP Addr>\<davRootDir>`
Also windows explorer does not support files in "\". Only folders which treated as "shared folders". The davRootDir can solve this trouble but windows explorer still trying to check the whole path.

Can all subdirectories within davRootDir path be reported as virtually exists?

d-a-v, 2023-01-17:
Can you please provide a way to reproduce this issue ?
I do not use and neither have an easy access to windows so I'd prefer have as much as input before trying.
Thanks

pipi61, 2023-01-17:
I don't understand, this works
[image]

nightgryphon, 2024-03-17:
@pipi61 put some files in to same DavWWWRoot located than try to open \\192.168.0.24\