d-a-v/ESPWebDAV #22: Compilation error

jajito, 2022-11-28:
HI, I have changet the 
#include <LITTLEFS.h>
to 
#include <LittleFS.h>
and every LITTLEFS to LITTLEFS, as sugested by the manteiner of the library.
I platformio no appears any error or problem, but when compiling it appears a problem in ESPWebDAV.cpp:

include/ESPWebDAV.cpp:1091:48: error: invalid conversion from 'char*' to 'const uint8_t*' {aka 'const unsigned char*'} [-fpermissive]

Am I doing something wrong or forgot something?.
Thanks ini advance

serisman, 2022-11-29:
Changing line 1091 from: `auto numWrite = file.write(buf + written, numRead - written);`
to: `auto numWrite = file.write((uint8_t*)buf + written, numRead - written);`
seems to fix the compile error.

Still testing if this actually works or not...

jajito, 2022-11-29:
Now it compiles!
Thanks serisman!

jajito, 2022-12-01:
But doesn't work.
Doesn't send info of the files or folders to the FTP client...
The rest works ok...

d-a-v, 2023-01-17:
Can you please give more input about your issue ?

d-a-v, 2023-01-18:
> Changing line 1091 from: `auto numWrite = file.write(buf + written, numRead - written);` to: 
> `auto numWrite = file.write((uint8_t*)buf + written, numRead - written);`

fixed by #23

closed 2023-07-16 by d-a-v